bremme/arduino-tm1637 #1: Decimal dot

silverchair, 2016-02-10:
Hello,
the library is pretty cool, but is it possible to dispaly decimal dot? Thanks in advance.

bremme, 2016-02-10:
Unfortunately, this is not possible. When sending a character to the display there is only one bit to set the colon (two dots), so there is no way to set only the lower dot of the colon. For more information see: SevenSegmentTM1637.h 

  // SevenSegmentTM1637 low level methods (use when you know what you're doing)
  /* Prints raw (encoded) bytes to the display
  *         A
  *       ___
  *  * F |   | B
  * X     -G-
  *  * E |   | C
  *       ___
  *        D
  * Bit:            76543210
  * Segment:  XGFEDCBA
  *
  * For example to print an H, you would set bits BCEFG, this gives B01110110 in binary or 118 in decimal or 0x76 in HEX.
  * Bit 7 (X) only applies to the second digit and sets the colon

Nice, that you like the library!:)

yucelll, 2016-07-12:
hello. this library the best library in the all tm1637. basic and fun!
But i dont show "dot"

and i am little known arduino.
i using lm35 and tm1637
i want see on the display: " 25.60 "

and later i want see " 25.6° "

please help me! thanks.

bremme, 2016-11-14:
I'm working on a major upgrade for the library which will have better support for the things you mention. I just tested with the [Robodyne](https://www.aliexpress.com/store/product/4-Digit-LED-Display-Tube-7-segments-RED/1950989_32628991527.html) displays and found out that these displays allow to set the lower and upper dot of the colon separately. The upper colon is set by setting bit 7 of digit 3 and the lower colon is set by setting bit 7 of digit 4. You should be able to set the colon's by using the `printRaw` function.

To print for example `25.60` on the TM1637 Robodyne display you would use:

```arduino
byte  rawData[4];
rawData[0] = display.encode('2');
rawData[1] = display.encode('5');
rawData[2] = display.encode('6') ;
rawData[3] = display.encode('0') | B1000000; // set the lower colon, bit 7 on digit 4
display.printRaw(rawData);
```

To print the degree symbol, you need to set segment A, B, G and F of the last digit. This can be done by:

```arduino
// same as previous example, only now we set the 4th digit to the degree symbol
rawData[3] = B01100011 | B1000000; // set the lower colon, bit 7 on digit 4
display.printRaw(rawData);
```

hectorlee, 2016-12-06:
I'm using the Robodyn display but it doesn't seem to be able to control the upper and lower colon separately. I'm using one with green 7 segment displays.

This is the code for controlling the colon on mine.

```byte  rawData[4];
rawData[0] = display.encode('2');
rawData[1] = display.encode('5') | B10000000; // turn on the colon
rawData[2] = display.encode('6') ;
rawData[3] = display.encode('0');
display.printRaw(rawData);
```

I'm actually most interested in the decimal point on the individual digits can be controlled. Do you think this is possible? The TM1637 should have sufficient pins to do so.

julienbrunet, 2016-12-07:
Same here, I bought those sepcific [RobotDyn ](https://fr.aliexpress.com/item/4-Digit-LED-Display-Tube-7-segments/32591091564.html?spm=2114.13010608.0.0.UvMZHG&detailNewVersion=&categoryId=541) displays because of their decimal dot
But after reading more carefully the description on aliexpress, it says the decimal dot is not active...

And i'm also not able to control the upper / lower colon separatly :(

bremme, 2016-12-08:
I believe the decimal dots are not wired to the chip on these RoboDyne displays. I have a red version at my desk and this one can display the upper and lower colon by setting the 7th bit (counting from zero). I noticed that I made a typo in my previous post (B1000000 vs B10000000). I just tested it again and it does work on my display:

[image]

This is the code I used:

```arduino
  byte  rawData;
  display.print("256");
  // display degree symbol on position 3 plus set lower colon
  rawData = B11100011;
  display.printRaw(rawData, 3);
```

hectorlee, 2016-12-08:
I guess there might be a difference in the hardware that RobotDyn makes. I loaded your code and this is what I get.

[image]

Are you able to control turning on both semi-colon with bit 7 of the second digit?  This is the code for turning both parts of the semi-colon.

```byte  rawData;
  display.print("256");
  // display degree symbol on position 3 plus set lower colon
  rawData = B11100011;
  display.printRaw(rawData, 3);
```

dterweij, 2017-01-18:
Hi, I moved to this library because it is a much  better (and cooler) one.
Just how to do a "27.3C"?
String temperature = "27.32"; (tho making it  a int is no problem)
Maybe using a rouding math for the 4th digit of my string? so .32 becomes .3 and .37 becomes .4 ?
And using the colon for a dot.

dterweij, 2017-01-18:
Got it on my own:
int disptemp = 29.09
newTemp makes it 29.1
then times 100 makes 291
set colon on and degree symbol

Hope it helps someone else too.

```
    float newTemp = disptemp / 10.0; // rounding decimal
    disptemp = newTemp * 100;    
    byte rawData;
    rawData = B01100011;
    display.setColonOn(1);
    display.print(disptemp);
    display.printRaw(rawData,3);
```

techi602, 2017-03-19:
I also thought that the robotdyne display features decimal points per digit, but unfortunetly not:

> This is a basic 4-digit 7-segment display module (GREEN). The display features double points in midle. **The decimal point per digit not active.**
> Module connecting to digital I/O on 2 pins.
> For Arduino use library: TM1637.h

http://robotdyn.com/catalog/segment/4_digit_led_display_tube_7_segments/

Highcooley, 2018-03-25:
The new ones do:
http://robotdyn.com/4-digit-led-display-tube-7-segments-tm1637-50x19mm.html

ybaransky, 2018-04-02:
Bram (or anyone) - how can you tell which of the TM1637 based 4-digit LED's are enabled for decimal point or colon functionality? I purchased 
https://www.ebay.com/itm/112674024638?ul_noapp=true
thinking it would support both. Looks like it only supports colon mode.  I was hoping there would be a way to set the board up in colon or dp mode via software. Does not look good. Thx -yurij

mrkale, 2018-04-03:
It is written in the name of the product "LED Clock Tube". On more serious pages it is exactly written in the product specification, that a decimal point is not active.
It is not necessary to vary a library. The colon is active just for active 7 segment (counting from 0, a.k.a. DP) of the second tube no. 1 (counting from 0). Decimal points of other tubes (digits) do not display even if their DP segment is activated. It is up to you to proper format the displayed data.

ybaransky, 2018-04-03:
Thx
1)  but the device I purchased
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F11267
4024638
does not explicitly say one or the other
2) the image describing the part shows both colon and decimal point enabled
3) clocks can also be stopwatchs and there tenths and even hundreds of
seconds are important
4) I have seen recently tm1637 parts that offer colon or decimal as a
different  part. Not software specifiable. So decimal point is possible.
5) any chance you know of a 4-digit, 7-segment display driver that can
light decimal and colon and is inexpensive? Adafruit has such parts at $10 USD
each. They use a different chip.

Separately, how did you email me directly? I have no notification on the
github site and can't seem to find the emails of people I want to speak
with without posting my questions on github

Thx for answering. It does not look promising for the parts I have.
-yurij

mrkale, 2018-04-03:
> ebay.com%2Fulk%2Fitm%2F11267

Hi, friend.

Ad 1) In contrary, it does. The description starts with "This module is a
12 foot with *clock display* four common anode tube".
Ad 2) Yes, the digital tubes on a display have usually technically present
decimal point and colon, but only one of them is broken out on the display
connector or connect to the TM1637. The best way is to find schema and
description of the display by a product number on its side, e.g.,
http://www.epeslight.com/led-7-segment-display/39514793.html.
Ad 3) I would recommend displays with TM1638, which are usually equiped
with 8 tubes with decimal points and are chainable.
Ad 4) Yes, for instance RoboDyn company offers display modules with TM1637
and either with colon or decimal points clearly distinguished:
https://robotdyn.aliexpress.com/store/group/Segment/1950989_504708861.html?spm=2114.12010612.0.0.694950392nSiUO
.
Ad 5) There are digital tubes display with separate controlled colon dots,
decimal dots, eventually the degree dot, e.g. aforementioned display
GNS-5643AxBx <http://www.epeslight.com/led-7-segment-display/39514793.html>.
The problem would be a controller, resp. a cheap module with all mentioned
dots controlled at once. I would recommend TM1638 or MAX7219 (7221) and to
consctruct display module on your own.

I did not email to you directly. I just add my contribution on the GitHub
issue page and the portal resent it to your registered address.

Best Regards
mrkale

ybaransky, 2018-04-03:
Very informative, thank you. Though I would still quibble with the term
clock.

I hope I can repay the favor. Knowledge is expensive to acquire.
-yurij

Chinoforest, 2018-07-26:
Hi. Any plans on supporting this?
https://robotdyn.com/4-digit-led-display-tube-7-segments-tm1637-30x14mm-red-decimal-points.html

av4625, 2018-07-29:
If you get the colon one from robotdy can you just turn on the bottom dot? Will you support the decimals one at some point?

heatvent, 2018-10-01:
Hi, another person asking if decimals will be supported?  If not, is it possible to use the TM1637Display library for decimal numbers (which it supports) and also use your library for the many other cool features!

heatvent, 2018-10-16:
Is there possibly a way to force decimals to work with this library.  I noticed the following in the sevensegmenttm1637.h file...

`#define TM1637_COLON_BIT        B10000000`

very similar to the TM1637Display.h library for the showdecimalex function...

```
> Dispalyes the given argument as a decimal number. The dots between the digits (or colon)
>   //! can be individually controlled
>   //!
>   //! @param num The number to be shown
>   //! @param dots Dot/Colon enable. The argument is a bitmask, with each bit corresponding to a dot
>   //!        between the digits (or colon mark, as implemented by each module). i.e.
>   //!        For displays with dots between each digit:
>   //!        * 0.000 (0b10000000)
>   //!        * 00.00 (0b01000000)
>   //!        * 000.0 (0b00100000)
>   //!        * 0.0.0.0 (0b11100000)
>   //!        For displays with just a colon:
>   //!        * 00:00 (0b01000000)
>   //!        For displays with dots and colons colon:
>   //!        * 0.0:0.0 (0b11100000)
>   //! @param leading_zero When true, leading zeros are displayed. Otherwise unnecessary digits are
>   //!        blank
>   //! @param length The number of digits to set. The user must ensure that the number to be shown
>   //!        fits to the number of digits requested (for example, if two digits are to be displayed,
>   //!        the number must be between 0 to 99)
>   //! @param pos The position least significant digit (0 - leftmost, 3 - rightmost)
>   void showNumberDecEx(int num, uint8_t dots = 0, bool leading_zero = false, uint8_t length = 4, uint8_t pos = 0);
```
It actually turns on the middle decimal.  I am bit / byte illiterate so I am wondering if there is a way to change the definition of TM1637_COLON_BIT or to send a raw command to control the decimal of choice.

Thank you

heatvent, 2018-10-16:
OK, I tracked down the code that controls the colon and basically the logic printraw changes a bit on whatever you are sending to the left of the colon position.  I forked the code at https://github.com/heatvent/arduino-tm1637 and added (hacked) some code to control the decimal.  Not sure if I have this working correctly and decimals really need some logic depending on if you want to show x number of decimal places, just one of the decimals to light up, have the decimal move with the number being shown (0.01 and 0.001 vs 0.010 and 0.001).  If something like this could get integrated into the master that would be great, otherwise this is working for me for now.

benladenusam, 2019-05-30:
> OK, I tracked down the code that controls the colon and basically the logic printraw changes a bit on whatever you are sending to the left of the colon position. I forked the code at [...]

DAAAAAMN, man, you made my day. THANKS!!!

JensGrabner, 2019-05-30:
You can find here ?  --  https://github.com/avishorp/TM1637

nuxeh, 2022-05-16:
Here is another lib with decimal support https://github.com/jasonacox/TM1637TinyDisplay