bremme/arduino-tm1637 #5: bombTimer not accepting 0 hours

fooons, 2016-10-29:
Hello bremme.

in the countdown bombtimer if I want to setup my code for only 30 seconds, it can't be possible and the counter never passes through the zero. For example, if I put 1 minutes and 5 seconds, when the seconds counter arrives to zero does not start in the 59 remaining seconds for the last minute.

Sequence example 01:05 - 01:04 - 01:03 - 01:02 - 01:01 - 01:00 - 00:00 (it should be 00:59) 

Thanks in advance.

bremme, 2016-11-02:
Been quite busy lately, sorry for the late response. I Will look into this soon and I will let you know when it's fixed.

fooons, 2016-11-02:
No problem, I understand you are a busy man. Thanks for your efforts

bremme, 2016-11-04:
I just had a look and solved this issue. If you want you can test the latest version to see if it works for you as well. I tested the countdown using the sketch below:

```arduino
// #include "SevenSegmentTM1637.h"
// #include "SevenSegmentExtended.h"
#include "SevenSegmentFun.h"

const byte PIN_CLK = 4;   // define CLK pin (any digital pin)
const byte PIN_DIO = 5;   // define DIO pin (any digital pin)
SevenSegmentFun    display(PIN_CLK, PIN_DIO);

// run setup code
void setup() {
  Serial.begin(9600);         // initializes the Serial connection @ 9600 baud
  display.begin();            // initializes the display
  display.setBacklight(100);  // set the brightness to 100 %
  delay(1000);                // wait 1000 ms
};

// run loop (forever)
void loop() {
  byte hours = 1; byte min = 16; unsigned int speed = 60;
  display.bombTimer(hours, min, speed);
  delay(1000);;
};
```

closed 2016-11-04 by bremme

fooons, 2016-11-05:
It works perfect! 

Now I will try to Stop,Pause and Resume the timer and the implementation of a live speed change of the timing by pushing buttons or a pot value.

Thanks and best regards.