bremme/arduino-tm1637 #6: startup

Amritpal101, 2016-11-08:
During the startup of the program,  ON segment blinks very fast,  can i change the speed and segment also?

bremme, 2016-11-08:
Good point. At present you can't change the text, its hard coded. (See line 187 of `SevenSegmentTM1637.cpp`). You could change the blink speed by editing the `SevenSegmentTM1637.h` file, you would have to look at line 32, there the default blink delay is set to 50 ms by:

```arduino
#define TM1637_DEFAULT_BLINK_DELAY    50      // ms
```

What would you suggest? I could edit the `begin()` function like this for example

```arduino
begin(char text[], bool blink, blinkSpeed);
```

Where both `text`, `blink` and `blinkSpeed` are optional. If you don't give a text, nothing will be displayed. If you give a text, the text will be displayed. If you also set the blink boolean, the text will blink and so on.

What do you think?

per1234, 2016-11-08:
I'd say just get rid of that blinking "ON" message code in `begin()` altogether. Just have `begin()` set the values of rows and cols. The blinking message serves no useful purpose, adds bloat to the library, and is blocking. If the user wants something to blink on startup they can just code it in their sketch.

bremme, 2016-11-08:
@per1234 I agree that it should not display and blink a message by default. I guess I put it in their a while ago just for fun and just forgot about it. You could still have it optional, but perhaps it better to demonstrate such a thing using any of the examples.

Amritpal101, 2016-11-08:
> On 8 ਨ

Yes. I this this is the perfect way and best option for the user also if you keep these functions optional.  Because sometimes during the coding and testing user face this function (on) at every time of uploading. So better it should be optional.   
 
Thanks
Sent from an iPhone

bremme, 2016-11-14:
In the end I completely removed the text and blinking options. Like @per1234  said, if people want such a thing this can be easily achieved using the `print()` and `blink()` methods.

closed 2016-11-14 by bremme