bremme/arduino-tm1637 #7: comStop() not Working

Panther2804, 2016-11-10:
I can't get  'comStop();' to work, i keep getting
errors, and when im using 
  'display.comStop();'
It just doesn't work.

bremme, 2016-11-10:
Could you maybe post your whole sketch or if its very big the relevant parts? Then I can try to reproduce your error and see if I can solve it.

Panther2804, 2016-11-12:
Sure thing
`#include "SevenSegmentTM1637.h"

int countdown = 100;
const byte PIN_CLK = 4;   // define CLK pin (any digital pin)
const byte PIN_DIO = 5;   // define DIO pin (any digital pin)
SevenSegmentTM1637    display(PIN_CLK, PIN_DIO);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);         //Need this for a Bluethooth Module, wich i can't use if theres all this informaition on the Port 
  display.begin();        
  display.setBacklight(50);  
  display.comStop(); //note the command isn't highlighted like it should be
  
  display.print("Waiting for Input");
  //now there would be somer Serial input, but if you look at the serial Monitor the Port is Spammed with information
  delay(1000); //delay instead of input
  display.print("    "); // clear Display
}

void loop() {
  // put your main code here, to run repeatedly:

if(countdown>=1){
countdown --;};
display.print(countdown);
delay(1000);

//now a Siren would beturned on, but you dont need this
}`

Btw: Your library is awesome and i really enjoy Working with it because it is so easy tho use but this Error is really annoying because i can't start Debugging before it works.
Sorry for my not presen't english and Programming skills. Im just a 14 YO boy from Germany just getting started with the Arduino. Thanks for your help.

bremme, 2016-11-12:
Thanks, you really don't need to call the `comStop()` method. This method is part of the advanced low level methods. These methods (`command()` which calls `comStart()`, `comWriteByte()`, `comAck()` and `comStop()`) are used to preform the low level communication with the `TM1637` display. For normal use they are not required.

Perhaps my description is not clear enough and you though that this command has anything to do with the `Serial` communication with the Arduino. This is not the case these methods are used for the communication (this is also serial communication) with the display.

Hope this helps, let me know if you have any further questions or if I did not understand your comment perhaps.

bremme, 2016-11-14:
@Panther2804 Did you manage to get it to work? If so I will close this issue.

bremme, 2016-11-18:
@Panther2804 I guess you managed to solve the problem with `comStop()`? I will close this issue now. If you did not manage to solve it, I can always reopen this issue.

bremme, 2016-11-18:
Thinking about it again, I think your problem might be cause by the fact that probably the version you downloaded has (by mistake) the debugging flag set in `SevenSegmentTM1637.h`. You can download the latest version of the library again, or look for the following line in `SevenSegmentTM1637.h`:

```arduino
#define TM1637_DEBUG                  false   // true for serial debugging
```

closed 2016-11-18 by bremme