bremme/arduino-tm1637 #10: DigiSpark

TVRleigh, 2017-02-06:
Hi,
If I use the example with a UNO it works fine but if I change the board to a DigiSpark, I get the following error.

C:\Users\Leigh\Documents\Arduino\libraries\SevenSegmentTM1637\src/SevenSegmentTM1637.h:134:10: error: conflicting return type specified for 'virtual size_t SevenSegmentTM1637::write(const char*)'

   size_t write(const char* str);

per1234, 2017-02-07:
Here's the problem:
https://github.com/digistump/DigistumpArduino/blob/master/digistump-avr/cores/tiny/Print.h#L74
```c++
 virtual void write(const char *str);
```
and also:
https://github.com/digistump/DigistumpArduino/blob/master/digistump-avr/cores/tiny/Print.h#L75
```c++
 virtual void write(const uint8_t *buffer, size_t size);
```
so the problem is the `digistump-avr:tiny` core's Print API is inconsistent with the `arduino:arduino` core's Print API in that those two versions of `write()` return `void`. I see the `digistrump-avr:pro` core doesn't have the issue so I guess it's intentional because of the limited memory of the ATtiny85.

It would be possible to fix this by using the preprocessor to change the function signatures:
```c++
#if defined(ARDUINO_AVR_DIGISPARK) && defined(__AVR_ATtiny85__)
```
I'm not sure if you consider it worth adding code to the library only to support a specific core.

TVRleigh, 2017-02-08:
Thank you for your quick response at finding the problem.

I've looked into changing it, but I think its currently above my skill level. but as the code work on the PRO, I think I'll just use this board instead as its not much bigger. esp as I can write all the rest of the code I need and test with a UNO.