bremme/arduino-tm1637 #11: Blinking on seconds

Its-Matra, 2017-03-18:
Am I just being really dumb or can you not get the colon to blink on seconds? It blinks every minute but not seconds (using the clock example and examined SevenSegmentTM1637.h and SevenSegmentExtended.h but can't find how to get the damn thing to blink every second).

I also noticed an error in the clock example too, when hitting 23:59 the whole thing will just freeze, you need to add an hour reset too.

```
  for ( ; hours < 24; hours++) {                // count hours   up to 24
    for ( ; minutes < 60; minutes++) {          // count minutes up to 59
      display.printTime(hours, minutes, true);  // display time
      delay(60000 / clockSpeed);                // clock delay ms
    }
    minutes = 0;                                // reset minutes
  }
  **hours = 0;**
}
```

AJMartel, 2018-02-05:
Thanks Its-Matra for the "hours" fix.
Unfortunately it does not work for me.
I am also looking to get the colon to blink every second

bremme, 2018-04-09:
@Its-Matra Damn, this is a late reply. Finally found some time to reply to some issues. Just tested my clock example again `ExtClock.ino` and I think it works as expected: run the clock at high speed from 14:39 to 23:59 and reset. I you remove the hour and minute initialization lines from the demo, you have to reset the hours yourself like you found out. You probably already fixed this, but here is an example to show HH:MM and blink every second:

```c++
void loop() {

  byte hours    = 14;                           // initialize hours
  byte minutes  = 39;                           // initialize minutes
  byte seconds  = 0;                            // initialize seconds

  for ( ; hours < 24; hours++) {                // count hours   up to 24
    for ( ; minutes < 60; minutes++) {          // count minutes up to 59
      for ( ; seconds < 60; seconds++) {
        display.printTime(hours, minutes, true);  // display time
        delay(1000);
      }
      seconds = 0;
    };
    minutes = 0;                                // reset minutes
  };
};
```

Hope this helps!

closed 2018-04-09 by bremme