bremme/arduino-tm1637 #12: Add support for TM1637 key/button input

timwaizenegger, 2017-04-09:
I've noticed that the TM1637 has two pins that you can use together with the LCD-grid pins as a key/button matrix. apparently you can connect up 16 buttons this way and read their state via the serial I/O.

It looks like very few off-the-shelve display modules for arduino/raspberry use this feature but it could be very useful in custom applications. When you're using a display it's most likely for a UI, so usually you'll want some buttons as well...

I found no arduino TM1637 lib that implements this feature. Your library seems very complete and ambitious; maybe you're interested in adding this  :+1:

timwaizenegger, 2017-04-09:
I did find a library that seems to implement the feature; might be useful as a reference.

https://forum.arduino.cc/index.php?topic=360318.0

bremme, 2017-04-10:
Hi Time, thanks for your feedback! There is already a low level method which should be able to read the key matrix (`uint8_t SevenSegmentTM1637::comReadByte(void)`). But I never really tested this since I didn't had a TM1637 display with connected buttons. I think I do have one now, if I can find it:). I'm going to have a look at this.