bremme/arduino-tm1637 #13: Add support for "4-Digit Display (D4056A) by Catalex"

rneppi, 2017-05-20:
I have bought LED-Displays with TM1637, Brand "4-Digit Display (D4056A) by Catalex". It seems, with your library displaying floats fails:
`display.print("abcd");` works fine
`display.print(82.123,0);` works fine. It is shown: "82  "
`display.print(82.123,1);` dont work. It is shown: "_0  "

By chance I noticed, that the provider has its own library. Look here: [https://drive.google.com/drive/folders/0B6uNNXJ2z4CxQjNKd2dkYU1QOFU?usp=sharing]

If possible and permissible I suggest, expanding your arduino-tm1637 library for these displays to. It would be great!

Product-Links
[https://www.aliexpress.com/store/product/D4056B-0-56-LED-4-Digit-Display-Module-with-Clock-Point-for-Arduino-AVR-ARM-PIC/1199788_2027331863.html?spm=2114.12010612.0.0.7ra7lO]
[https://www.aliexpress.com/store/product/0-36-4-Digit-LED-Display-Module-with-Decimal-Point-TM1637-Drive-Mini-7-Segment-LED/1199788_32648785549.html?spm=2114.12010612.0.0.xrKE6U]

bremme, 2017-05-30:
Sorry for my late reply. I already started to make the library much more modular, eventually it should support any seven segment display (at least that's the plan). Hopefully I can find some time to work on the library for a couple days the coming weeks to finish this. I will order the display you mentioned for testing.

For the time being, you could probably work around most issues using some of the advanced or low level methods.

rneppi, 2017-05-30:
I’m happy to read from you. It's great that you will checkout these displays to. Good luck!

By reading my post, I saw, that the link to the 0,56’’-Display is not, what I bougt. This 0,56’’-Display I bought: https://www.aliexpress.com/store/product/Free-Shipping-3Pcs-0-56-LED-4-Digit-Display-D4056A-4-Bits-Digital-Tube-LED-Module/1199788_1929888407.html
How ever, there are several Displays. Maybe interesting to you too: https://www.aliexpress.com/store/group/LED-LCD-Display-Module-TFT/1199788_259281354.html

Best regards
Rainer