bremme/arduino-tm1637 #15: Display empty on ESP32

jwktje, 2017-09-25:
I'm using this library on an ESP32 based module (Wemos LOLIN32 Lite) and nothing shows up on the screen. 

I've verified that the tm1637 screen works. Also verified my sketch on another microcontroller (Arduino UNO). The included example doesn't run on my ESP32 controller.

Any idea what the problem would be?

bremme, 2017-09-28:
I have not tested the library on a ESP32, I did not even know the ESP32 was Arduino compatible already. Could you post me the sketch and the compilation output? Perhaps that makes things more clear.

jwktje, 2017-09-28:
Hey man! Thanks so much for responding. I've been testing so much and updating in an issue on the ESP32 repo. 
https://github.com/espressif/arduino-esp32/issues/668

Basically I've excluded these possible causes:
- Difference in voltage to the screen
- Sketch errors (Code runs fine on Arduino Uno)
- ESP based incompatibility (that is; ESP8266 works fine)
- Pin choice (I've tried every single pin on my board)
- Difference in Clock speed (I upped the clock delay but to no avail)

The person that sold me the parts also helped me out a bit and checked mine (and your code) and suggested it might be a problem with the INPUT_PULLUP pinmode. 

Some info. When running your LowLevel example Sketch, 
## This is the compiler output:
```
Sketch uses 140,630 bytes (10%) of program storage space. Maximum is 1,310,720 bytes.
Global variables use 11,564 bytes (3%) of dynamic memory, leaving 283,348 bytes for local variables. Maximum is 294,912 bytes.
esptool.py v2.1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 2172.4 kbit/s)...
Hash of data verified.
Flash params set to 0x022f
Compressed 10560 bytes to 7084...

Writing at 0x00001000... (100 %)
Wrote 10560 bytes (7084 compressed) at 0x00001000 in 0.6 seconds (effective 130.2 kbit/s)...
Hash of data verified.
Compressed 141776 bytes to 78102...

Writing at 0x00010000... (20 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (60 %)
Writing at 0x0001c000... (80 %)
Writing at 0x00020000... (100 %)
Wrote 141776 bytes (78102 compressed) at 0x00010000 in 7.0 seconds (effective 162.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 122...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.0 seconds (effective 674.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

```
## This is the Serial output
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (Sets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:572
load:0x40078000,len:0
load:0x40078000,len:9880
entry 0x400789d8

0
0
0
0
0
0
0
0
0
0
```
It repeats those zeroes forever. So the write is not acknowledged.

bremme, 2017-10-09:
I'm sorry I have not found the time to look into this, I will try to respond ASAP. Or did you already found a solution yourself?

dunklesToast, 2018-03-18:
Has the same problem today. The ESP8266 does not have enough power for the board. I used 5V from an external power supply and it worked.

bremme, 2020-06-22:
Its been awhile that I had some time to work on this repo. I don't have an an ESP32 myself so I can't test this. I'm gonna close this for now. Please reopen if the problem still exists and or if you found a solution/work around.

closed 2020-06-22 by bremme