bremme/arduino-tm1637 #16: How to disable logging with Serial?

stpoa, 2017-12-05:
Hi,

I need serial output to send specific data, how to disable logging when you are displaying something on a screen?

bremme, 2017-12-05:
Hi @stpoa the library should not output any serial data, unless you have set:

```c++
// COMPILE TIME USER CONFIG ////////////////////////////////////////////////////
#define TM1637_DEBUG false // true for serial debugging
```

To `true` in the SevenSegmentTM1637.h header file. I did a quick search for the `Serial.println()` command and found that there is one method in the extended class (printDualCounter) where I left two instance by mistake. I will fix this asap. Are you using that method, or if not, when and what is printed over serial?

stpoa, 2017-12-05:
You are right, I was using Serial.println(). I cannot check the output at the moment, but it would be awesome if you could fix this. Thank You :D

bremme, 2017-12-05:
I'm not sure if I understand you properly. Are you using the `printDualCounter()` method or are you using `Serial.println()` yourself somewhere in your code. And what do you mean by: 'I cannot check the output at the moment'. Do you mean you get no text on for example the Arduino IDE serial monitor when you are using `Serial.println()`? In the latter case I don't think it has anything to do with the `SevenSegmentTM1637` library.

bremme, 2017-12-09:
Have you been able to solve your problem? I will close this issue for now.

closed 2017-12-09 by bremme