bremme/arduino-tm1637 #17: Count Down Problem

Persp0lisy, 2018-01-16:
hi and thanks a lot for this very very useful library.
i used display.print() function in my sketch and i have problem for counting down from 10 to 0
the display shows this:(10-90-80-70-60-50-40-30-20-10-00).
i cant not use display.clear() function because i am using degree sign it will be cleaned if i use clear function.
could you help me please?

bremme, 2018-01-17:
If you post your code/sketch here I can have a took and help perhaps.

Amritpal101, 2018-01-18:
May i have a look your code plz. 

Sent from an iPhone

Persp0lisy, 2018-01-21:
Here is my code:

#include "SevenSegmentTM1637.h"
const byte PIN_CLK = 14;   // define CLK pin (any digital pin)
const byte PIN_DIO = 15;   // define DIO pin (any digital pin)
SevenSegmentTM1637    display(PIN_CLK, PIN_DIO);

void setup() {
display.begin();
display.setBacklight(100);
byte  rawData[4];
//rawData[0] = display.encode('2');
//rawData[1] = display.encode('5');
//rawData[2] = display.encode('6') ;
rawData[3] = display.encode('c') | B10000000; 
display.printRaw(rawData);
};

void loop() {
 int input = analogRead(A5);
  int Temp = map(input , 0, 1023, 0 , 100);
  if (Temp <= 50){
  display.print(Temp);
  }
  if (Temp > 50){
    display.print(Temp);
    display.blink(500);}  

  }

BigRedBeard86, 2018-03-01:
I also am encountering this problem, down count shows: 13, 12, 11, 10, 90, 80, 70, 60, 50, 40, 30, 20, 10
I am trying to figure out how to turn off that column display while it is less than a 10 count. anyone got any luck with this?

BigRedBeard86, 2018-03-01:
Sorry for posting back to back, but I found a work around that "works"
[Down_Count.txt](https://github.com/bremme/arduino-tm1637/files/1773103/Down_Count.txt)

Persp0lisy, 2018-07-23:
hi again
that way is not cool,i think this problem should solved in library sorce