bremme/arduino-tm1637 #20: Multiple Displays.

BigRedBeard86, 2018-03-01:
Using your library on a MEGA 2560 board and works great.  I am trying to find a way to drive multiple displays with different displays on them.  I tried changing the output pins before each section of code for each display but it still only displays on one 4-digit display.  Is there any way to make multiple instances in the display and just call on those displays separately?

bremme, 2018-03-02:
I think it should just work if using two instances bound to different pins and displays. In your code you would just initialize two displays, something like this:

```c++
// Initialize display 1
const byte PIN_CLK_D1 = 4;   // define CLK pin (any digital pin)
const byte PIN_DIO_D1 = 5;   // define DIO pin (any digital pin)
SevenSegmentTM1637 displayOne(PIN_CLK_D1, PIN_DIO_D1);

// Initialize display 2
const byte PIN_CLK_D2 = 6;   // define CLK pin (any digital pin)
const byte PIN_DIO_D2 = 7;   // define DIO pin (any digital pin)
SevenSegmentTM1637 displayTwo(PIN_CLK_D2, PIN_DIO_D2);
```

Let me know if this helps.

BigRedBeard86, 2018-03-02:
YES! thank you very much for the timely response, this worked! I must appreciate the help!

closed 2018-04-09 by bremme