bremme/arduino-tm1637 #21: Decimal point not working

hannesfischer, 2018-03-30:
Hello,
how do I turn on decimal points?

Thank you

ybaransky, 2018-04-01:
same question. i also tried using avishort's tm1637 driver and i could only get the colon to work. no decimals. i am using the wemos d1 esp8266 to drive this (if it makes any difference). 
thx -yurij

bremme, 2018-04-09:
Some displays have not everything (dots/colons) wired. Could you guys post me an image or link to the display you are using. Furthermore, usually the dots/colons are connected to the last bit. Perhaps you guys can also figure out if and which dots/colons are connected to the 7th bit at which position:

```c++
  display.clear();
  for (unsigned char i=0; i < 4; i ++) {
    unsigned char lastBit = B10000000; // 128 in decimal, only enable bit 7
    display.printRaw(lastBit, i);
    Serial.print("Enable last bit at position: " ); Serial.println(i);
    delay(1000);
    display.clear();
  }
```

bremme, 2018-08-10:
I need to think of a better solution to this problem. It's all related to the different wiring and different displays used in combination with the TM1637. I will close this for now.

closed 2018-08-10 by bremme