bremme/arduino-tm1637 #22: Align Right?

dan2600, 2018-03-31:
Any way to make the digits align right instead of left when they are shorter than 4 digits?

bremme, 2018-04-09:
This sounds like a good option, have to think about a proper solution. The library just implements the Arduino print/write methods and these don't support padding or right alight. I should probably just add a custom method `printNumber` with a align argument. In the mean time you can just use a custom function yourself, quick and dirty example:

```c++
void printAlignRight(int number) {
  if (number < 10) {
    display.setCursor(0, 3);
  }
  else if ( number < 100) {
    display.setCursor(0, 2);
  }
  else  if (number < 1000) {
    display.setCursor(0, 1);
  } else {
    display.setCursor(0, 0);
  }
  display.print(number);
}
```

Hope this helps.

Bilick88, 2018-09-11:
Let me "improve" the example that bremme has showed:

```
void printRight(int number) {
  if(number>9999) {
    display.print("----");
  }
  else {
  if (number < 10) {
    display.setCursor(0,0);
    display.print(" ");
    display.setCursor(0,1);
    display.print(" ");
    display.setCursor(0,2);
    display.print(" ");
    display.setCursor(0,3);
  }
  else if ( number < 100) {
    display.setCursor(0,0);
    display.print(" ");
    display.setCursor(0,1);
    display.print(" ");
    display.setCursor(0,2);
  }
  else  if (number < 1000) {
    display.setCursor(0,0);
    display.print(" ");
    display.setCursor(0, 1);
  } else {
    display.setCursor(0, 0);
  }
  display.print(number);
  }
}
```

Hope it helps :)

dan2600, 2018-12-19:
Hey sorry to take so long to join in. Thanks for addressing this. I came up with a similar solution to Bilick88's. Basically adding in space padding if the digits are shorter than a certain length. 

I'm not really very good at C optimization, so I'm not sure what is faster. But in my case I casted the int to a string and added spaces in a  (While len < 4) loop.

jmlcatarino, 2020-03-10:
Hi there, has anyone got a solution for this issue? I've tested the above solutions and it changes nothing to my sketch. 
I'm trying to create a parts counter but with this thing on the left it just doesn't look right.

jasonacox, 2020-06-14:
The example functions above do not handle negative values.  I made a few minor tweaks:

```c
void printNum(int number) {
  bool negative = false;

  display.clear();
  if (number < 0) {
    number = -number;
    if (number >= 1000) {
      display.print("----");
      return;
    }
    negative = true;
  }
  if (number > 9999) {
    display.print("----");
  }
  else {
    if (number < 10) {
      display.setCursor(0, 3);
    }
    else if ( number < 100) {
      display.setCursor(0, 2);
    }
    else  if (number < 1000) {
      display.setCursor(0, 1);
    } else {
      display.setCursor(0, 0);
    }
    display.print(number);
  }
  if (negative) {
    display.setCursor(0, 0);
    display.print("-");
  }
}
```