bremme/arduino-tm1637 #27: Fun library crashes board

runvnc, 2018-07-20:
My device has this printed on it: 3461BS1-7.3.

I wanted to try out the nightrider() thing but every time I try to include the Fun library stuff a weird sort of random series of lines displays and then everything seems frozen.  I can't even program it again until I unplug my shield with the LED stuff.

But the basic library works fine so I will stick with that.  Guessing maybe mine is an older model that can't do some fancy stuff maybe.

bremme, 2018-08-10:
Which Arduino board are you using and could you post me your sketch (the one that crashes) here?

bremme, 2018-09-29:
Hi @runvnc I'm gonna close this issue. Let me know if you can tell me a little more on your issue, then I will reopen this.

closed 2018-09-29 by bremme