bremme/arduino-tm1637 #29: Not treating negative numbers correctly

Chinoforest, 2018-08-30:
Hello. I was stydying your library to see if I could adapt it to work with TM1637 modules with decimal dot and I've found out it does not handle well floats. 

For example. 
If I print -1 (as int) the display shows correctly -1
If I print "-1" (as string) the display shows correctly -1
But if I print -1.0 (as float) the display shows -_0 

I'm looking at the Print.cpp file and at your library but I yet have to figure out where the error is. But there's definitely an error.

Anyway, thanks for this amazing library. I hope to improve it for decimal dots but first I have to undestand it!! haha

Chinoforest, 2018-08-30:
It looks that the problem comes from the way you implemented the write functions. In one write function you use the buffer _rawBuffer but in the others you don't. So when the Print function mixes both write functions, this problems arise.

For example:
	display.print('h');
	display.print('o');
	display.print("l");
	display.print('a');
	while(1);

Prints "hola" on the display.

But:

	display.print('h');
	display.print('o');
	display.print("-");
	display.print('a');
	while(1);

prints "hoa". Because the string function does not stores the printed chars on _rawBuffer and does not increment the cursor.

bremme, 2018-09-29:
Hi @Chinoforest thanks for looking in to this problem! You are more then welcome to make a pull request. Always too much things to do with too little time. I will try to solve this issue when I find some time to improve the library. In the meantime, perhaps I should add some current limitation to the README, there are a couple more things which don't work the way I like.