bremme/arduino-tm1637 #31: Publish to PlatformIO

6wheels, 2018-12-19:
Hi,

I use PlatformIO with VSCode to easily build and manage my projects.
It would be great if you could publish your lib to it as well.
I see that your project already has a library.json file, so it should be straight forward.
Here is a link to the platformio docs for publishing: https://docs.platformio.org/en/latest/librarymanager/creating.html

Thanks.

per1234, 2018-12-20:
This library has a library.properties file, not library.json. But that's fine because [PlatformIO also supports library.properties as a library manifest](https://docs.platformio.org/en/latest/userguide/lib/cmd_register.html#description).

vazquezjm, 2019-07-22:
Done!
[image]

6wheels, 2019-07-23:
Great! Thanks!

closed 2019-07-23 by 6wheels