bremme/arduino-tm1637 #32: Rotary Encoder and Display

gitandrewd, 2019-04-03:
Hi,

Firstly thanks for this great Library!

I am currently using it with Catalex TM1637 with a rotary encoder. The issue I am having is that every time I scroll up from 90 pass 100 then down again below 99, the third segment remains at 0. So 99 and anything below becomes 990. Is there any way for me to clear than third segment once it drops below 99.

Thanks for your time!
Dave

Amritpal101, 2019-04-03:
You can use it with this logic   
  

  
If (value  <=99){
  
  
display.setCurser(0,0);
  
displayPrint(value);
  
  
display.setCurser(0,2)
  
displayPrint(“    ”);
  
}else if(value  >=100){
  
display.setCurser(0,0);
  
displayPrint(value);
  
}
  
  
  
  
  
  
  
  Amritpal Singh

gitandrewd, 2019-04-03:
Worked Perfect! Thanks Amritpal for the quick reply :)

closed 2019-04-03 by gitandrewd