bremme/arduino-tm1637 #34: blinking the digits

shoaibkazemi, 2019-06-13:
my digits are blinking
how can i fix it to show fix digits

shoaibkazemi, 2019-06-13:
my code is :

#include <Wire.h> 
#include <Sodaq_DS3231.h> 
DateTime now; 
#include "SevenSegmentTM1637.h"
#include "SevenSegmentExtended.h"
#include <TM1637Display.h>
#include "JDateLib.h"
String jdate; // Jalali Date String
int JY,JM,JD; 
int check;  
int s,s1,s2,s3,s4,s5;
int z,z1,z2,z3,z4,z5;
int a,a1,a2,a3,a4,a5;
int m,m1,m2,m3,m4,m5;
int e,e1,e2,e3,e4,e5;
const byte PIN_CLK = 5;   // define CLK pin (any digital pin)
const byte PIN_DIO = 4;   // define DIO pin (any digital pin)
SevenSegmentExtended      displaysobh(PIN_CLK, PIN_DIO);
SevenSegmentExtended      displayzohr(7, 6);
SevenSegmentExtended      displayasr(9, 8);
SevenSegmentExtended      displaymaghreb(19, 18);
SevenSegmentExtended      displayesha(21, 20);

#include <SPI.h>
#include <SD.h> 
File myFile; 
void setup()
{
 pinMode(10, OUTPUT);
 pinMode (1,OUTPUT);//ettesale buzzer be paye 1 ya txd arduino
 displaysobh.setBacklight(100);
 displayzohr.setBacklight(100);
 displayasr.setBacklight(100);
 displaymaghreb.setBacklight(100);
 displayesha.setBacklight(100);
 displaysobh.begin();
 displayzohr.begin();
 displayasr.begin();
 displaymaghreb.begin();
 displayesha.begin();

 Serial.begin(9600);
 Wire.begin();
 rtc.begin();
 delay(5000);//Wite For Connect To Serial Port
 SD.begin(10); 
  
//DateTime dt(2019,2,19, 23, 51, 30, 3);
// rtc.setDateTime(dt);
}
  void oghat(int count,int h,int m)
  {
  int i;
  int check=(h*100)+m;  
  myFile = SD.open("oghat/oghat.txt", FILE_READ);
    while (myFile.available())
    {
        for (i=1;i<=count-5;i++)
          String temp = myFile.readStringUntil('\n');
        s1 = myFile.parseInt();
        s2 = myFile.parseInt();
        s3 = myFile.parseInt();
        s4 = myFile.parseInt();
        s=(s1*1000)+(s2*100)+(s3*10)+s4;    
        z1 = myFile.parseInt();
        z2 = myFile.parseInt();
        z3 = myFile.parseInt();
        z4 = myFile.parseInt();
        z=(z1*1000)+(z2*100)+(z3*10)+z4;
        a1 = myFile.parseInt();
        a2 = myFile.parseInt();
        a3 = myFile.parseInt();
        a4 = myFile.parseInt();
        a=(a1*1000)+(a2*100)+(a3*10)+a4;
        m1 = myFile.parseInt();
        m2 = myFile.parseInt();
        m3 = myFile.parseInt();
        m4 = myFile.parseInt();
        m=(m1*1000)+(m2*100)+(m3*10)+m4;
        e1 = myFile.parseInt();
        e2 = myFile.parseInt();
        e3 = myFile.parseInt();
        e4 = myFile.parseInt();
        e=(e1*1000)+(e2*100)+(e3*10)+e4;
        displaysobh.printTime(s,true);
        displayzohr.printTime(z,true);
        displayasr.printTime(a,true);
        displaymaghreb.printTime(m,true);
        displayesha.printTime(e,true);
        if (check==s) 
        {
            digitalWrite(1,HIGH);
            delay (500);
            digitalWrite(1,LOW);
            for (int i=1;i<=5;i++)
            {
              delay(1000);
              displaysobh.clear();
              delay(1000);
              displaysobh.printTime(s,true);
              
            }

            delay(50000);
        }
        if (check==z) 
        {
            digitalWrite(1,HIGH);
            delay (500);
            digitalWrite(1,LOW);
            for (int i=1;i<=5;i++)
            {
              delay(1000);
              displayzohr.clear();
              delay(1000);
              displayzohr.printTime(z,true);
              
            }
            delay(50000);
        }
        if (check==a) 
        {
            digitalWrite(1,HIGH);
            delay (500);
            digitalWrite(1,LOW);
            for (int i=1;i<=5;i++)
            {
              delay(1000);
              displayasr.clear();
              delay(1000);
              displayasr.printTime(a,true);
              
            }
            delay(50000);
        }
        if (check==m) 
        {
            digitalWrite(1,HIGH);
            delay (500);
            digitalWrite(1,LOW);
            for (int i=1;i<=5;i++)
            {
              delay(1000);
              displaymaghreb.clear();
              delay(1000);
              displaymaghreb.printTime(m,true);
              
            }
            delay(50000);
        }
        if (check==e) 
        {
            digitalWrite(1,HIGH);
            delay (500);
            digitalWrite(1,LOW);
            for (int i=1;i<=5;i++)
            {
              delay(1000);
              displayesha.clear();
              delay(1000);
              displayesha.printTime(e,true);
              
            }
            delay(50000);
        }
        
      
        break;
    }
    myFile.close();
}
void loop()
{

  now = rtc.now();
  int h = now.hour();                           
  int m = now.minute();
  int y = now.year();
  int mo = now.month();
  int d = now.date();
  Date now(y, mo,d);
  jdate = now.JDate();
  JY = now.JYear();
  JM = now.JMonth();
  JD = now.JDay();
  int mahrooz = (JD*100)+JM;
 // if (check!=mahrooz)
 // {
    
  switch (JD)
  {
    case 1:
      oghat (JM*5,h,m);  
      break;
    case 2:
      oghat ((JM+31)*5,h,m);
      break;
    case 3:
      oghat ((JM+62)*5,h,m);  
      break; 
    case 4:
      oghat ((JM+93)*5,h,m);
      break;
    case 5:
      oghat ((JM+124)*5,h,m);
      break;
    case 6:
      oghat ((JM+155)*5,h,m);
      break;
    case 7:
      oghat ((JM+186)*5,h,m); 
      break; 
    case 8:
      oghat ((JM+216)*5,h,m);
      break;
    case 9:
      oghat ((JM+246)*5,h,m);
      break;
    case 10:
      oghat ((JM+276)*5,h,m);
      break;
    case 11:
      oghat ((JM+306)*5,h,m);
      break; 
    case 12:
      oghat ((JM+336)*5,h,m);
      break;
 // }
}
  check=mahrooz;
}

bremme, 2019-06-13:
You already opened this issue, I will reply the original one and close this one.

closed 2019-06-13 by bremme