bremme/arduino-tm1637 #38: Drive two (or more) TM1637 displays from this library?

RonCWalker, 2020-02-13:
I wish to use two 4x7-seg displays, both TM1637. Is it possible to define the CLK and DIO pins for each with this library?

Cesar-S, 2020-02-13:
yes ...
SevenSegmentTM1637    display_1( 4 ,  5);

SevenSegmentTM1637    display_2( 7 ,  8);

RonCWalker, 2020-02-13:
Wonderful, many thanks for quick response.Regards, Ron

Sent from Yahoo7 Mail on Android

bremme, 2020-02-13:
@Cesar-S thanks for helping @RonCWalker! I'm not sure if I ever tested this, but it should just work if you just initialize two display objects like you showed. @RonCWalker I will close this issue for now, if you run into any issues I can always reopen this.

closed 2020-02-13 by bremme

RonCWalker, 2020-02-13:
Thanks for your help.I'm still weeks away from getting the parts to test this, but will let you know the outcome.Kind regards, Ron.

RonCWalker, 2020-03-04:
Hello Bram,Many thanks for taking this enquiry, unfortunately I have struck a problem, I believe.I was about to assemble the hardware, and started working on the code, just editing an existing simple example.In the line creating the object, "SevenSegmentTM1637 display(clkPin, dioPin)" - as soon as I enter anything other then "display" as the object, the colour of the word turns from a happy orange to black - this would suggest the text "display_1" was not acceptable.I persisted using the BASIC sketch in the example folder, duplicating the code creating the object;
const byte PIN_CLK1 = 4;   // define CLK pin (any digital pin)const byte PIN_DIO1 = 5;   // define DIO pin (any digital pin)
SevenSegmentTM1637 display_1(PIN_CLK1, PIN_DIO1);
const byte PIN_CLK2 = 6;   // define CLK pin (any digital pin)
const byte PIN_DIO2 = 7;   // define DIO pin (any digital pin)
SevenSegmentTM1637 display_2(PIN_CLK2, PIN_DIO2);
What I found was the program appeared to only recognise the first declaration of the object, in this case "display_1" and would not perform any display of "display_2"I tested this several times by reversing the names and it was always the first one listed that worked.

Do you know if I can modify anything to change this?Please see the attached code - I have massaged this several times without success.
Don't give this a high priority as I will be out of internet connection again for the next 12 days or so.

Kind regards, Ron

Cesar-S, 2020-03-04:
Hello, you must start the two displays, as if they were two different libraries.

Then you choose what to show and what to delete in each one.[image]

RonCWalker, 2020-03-04:
Thanks, I slept on the problem and concluded that would be the way forward.I appreciate your time on this.Regards, Ron

Cesar-S, 2020-03-04:
the orange color may be due to this ... display (), noDisplay (), https://www.arduino.cc/en/Tutorial/LiquidCrystalDisplay

bremme, 2020-06-22:
@Cesar-S yes, the reason that `display` shows orange in de Arduino IDE is because its a keyword. @RonCWalker so nothing to worry about it you use different names for the displays.