bremme/arduino-tm1637 #41: TM1637

PetitMalin, 2020-06-20:
Hi, 

Your library hes perfect for me, great job.
I used it with a 6 digits 7 segments from RObotDyn  just changed the line in file SevenSegmentTM1637.h
`#define TM1637_MAX_COLOM    6`
Works fine on USB.

But now i will use the library RS485 to communicate between a Mega2560(Master) and lot's of Nanos(Salves) with chipsets Max487

Since i changed to use the RS485 protocol during the compilation he gives me this error below:

`HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_18'
sketch\Radio_uhf_tm1637_RJ45.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_19'
sketch\Radio_uhf_tm1637_RJ45.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Erreur de compilation pour la carte Arduino Nano
`
Is it the part of the debug code ? ( Serial.begin and Serial.print ) in SevenSegmentTM1637.h 
It can be the communication that conflicts with RS485 protocol ?

Changed the `#define TM1637_DEBUG ` to false => but same result

Do you have a idea how to fix this ? 
Thank's for your help 

Ps: This is the code on USB to use with RS485, replace the line : 
`#define DCSBIOS_DEFAULT_SERIAL`
by 
`#define DCSBIOS_RS485_SLAVE 1
#define TXENABLE_PIN 2`
 

`#include <SevenSegmentTM1637.h>
#define TM1637_MAX_LINES 1 // number of display lines
#define TM1637_MAX_COLOM 6 // number of coloms (digits)
SevenSegmentTM1637 uhf(6,7);
SevenSegmentTM1637 vhf(4,5);
#define DCSBIOS_DEFAULT_SERIAL
#include <DcsBios.h>

void onUhfFrequencyChange(char* newValue) {
//Display from left to the right
uhf.setCursor(0,2);
uhf.print(newValue[0]);
uhf.setCursor(0,1);
uhf.print(newValue[1]);
uhf.setCursor(0,0);
uhf.print(newValue[2]);
uhf.setCursor(0,5);
uhf.printRaw(128,5);
uhf.setCursor(0,4);
uhf.print(newValue[3]);
uhf.setCursor(0,3);
uhf.print(newValue[3]);
}
DcsBios::StringBuffer<5> uhfFrequencyBuffer(0x7334, onUhfFrequencyChange);

void onVhfFrequencyChange(char* newValue) {
vhf.setCursor(0,2);
vhf.print(newValue[0]);
vhf.setCursor(0,1);
vhf.print(newValue[1]);
vhf.setCursor(0,0);
vhf.print(newValue[2]);
vhf.setCursor(0,5);
vhf.printRaw(128,5);
vhf.setCursor(0,4);
vhf.print(newValue[3]);
vhf.setCursor(0,3);
vhf.print(newValue[3]);
}
DcsBios::StringBuffer<5> vhfFrequencyBuffer(0x732a, onVhfFrequencyChange);

void setup() {
DcsBios::setup();
uhf.init();
uhf.clear();
uhf.printRaw(128,5);

vhf.init();
vhf.clear();
vhf.printRaw(128,5);
}

void loop() {
DcsBios::loop();
}`

bremme, 2020-06-22:
Good to hear that my library is useful to you. I did add some debugging functionality to the library which can be enabled by setting `TM1637_DEBUG` to `true`. But this is set to `false` by default. If you want to use the default hardware serial port on a Uno, Nano, Mini etc. to talk to a RS-485 IC you can't use the debug functionality of the library at the same time of course.

I'm not totally sure why you get these compilation errors but if you leave `TM1637_DEBUG` set to `false` I think it probably has nothing to do with the `SevenSegmentTM1637` library.

Do you get the same error if you don't use the `SevenSegmentTM1637` library?

I'm gonna close this for now since I don't think it a bug in the `SevenSegmentTM1637` library.

closed 2020-06-22 by bremme

PetitMalin, 2020-06-22:
Thx for your reply, unfortunately in my case i have always the error with SevenSegmentTM1637 and RS-485 library...

I tried without the SevenSegmentTM1637 library no error at compilation 

I also tried with a another TM1637 library (exemple this one https://github.com/AKJ7/TM1637) 
No error at compilation (but with this library i can not choose a digit in particular and this is essential for my project to have a result...) 

And no change to turn the debug false or true, always the same error at compilation   

I also tested other TM1637 library but no chance to have the possibility to choose a digit like your library...
So they seem to be going in a circle since 15 days.... Grrrrrrrr

bremme, 2020-06-22:
Okay, maybe I can still help you. Can you send me the full compile error and sketch.

PetitMalin, 2020-06-22:
Great Thx for your help 👍 

Here is attached the 2 sketchs named RJ45 for the RS485 protocol  
and USB to use with Nano connected on USB 
The difference between the two sketch :
RS485 protocol use: 
 `#define DCSBIOS_RS485_SLAVE 1
#define TXENABLE_PIN 2`
And USB: 
`#define DCSBIOS_DEFAULT_SERIAL`
And (if you need) this is the link for the DCS library : https://github.com/dcs-bios/dcs-bios-arduino-library 

Finally the error: 

`

HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':

(.text+0x0): multiple definition of `__vector_18'

sketch\Radio_uhf_tm1637_RJ45.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':

(.text+0x0): multiple definition of `__vector_19'

sketch\Radio_uhf_tm1637_RJ45.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Erreur de compilation pour la carte Arduino Nano
`
 
Again Thx (a lot)
At your disposal if needed

[Radio_uhf_tm1637_RJ45.zip](https://github.com/bremme/arduino-tm1637/files/4814334/Radio_uhf_tm1637_RJ45.zip)
[Radio_uhf_tm1637_USB.zip](https://github.com/bremme/arduino-tm1637/files/4814325/Radio_uhf_tm1637_USB.zip)

 ;)

bremme, 2020-06-22:
Just tested both your programs and they compile fine here:

[image]

[image]

bremme, 2020-06-22:
Are you sure that was the full error message you got? It has something to do with the HardwareSerial but I'm not sure what's going on. Something must be wrong with your configuration?

PetitMalin, 2020-06-22:
This is the log by default (Preferences/compiler warnings) : 
[7seg.log](https://github.com/bremme/arduino-tm1637/files/4815338/7seg.log)
 
Can you please post your log from the Radio_uhf_tm1637_RJ45

I will try tomorrow with a another computer, but it's weird...

PetitMalin, 2020-06-23:
Something is very very weird... 

I try on a new OS Ubuntu in a VM and i have the same error...
Ask on a friend to try, same result 

VM Ubuntu:
[image]

Here is the complete log on my VM Ubuntu: 
[ubuntu.log](https://github.com/bremme/arduino-tm1637/files/4821497/ubuntu.log)

Do you have a different version of your library SevenSegmentTM1637 ?

bremme, 2020-06-25:
Have you managed to find the problem? I'm working on some improvements and
I noticed that there were some `Serial.println()` calls in printDualCounter.
These will be removed in the upcoming version. But perhaps that related to
you issue? You could try to remove those and see what happens.

PetitMalin, 2020-06-28:
I'm not sure to understand what you asking me.
You asking me to remove `Serial.println()` in the SevenSegmentTM1637.h file ?
But if i change the debugging part (full or some lines), i have a another error during compilation 
(obviously linked to the modification).
Can you bring a little more details on the modification to make ?
THX in advance

bremme, 2020-06-28:
No, I was referring to the printDualCounter methods implementation in the
SevenSegmentExtended.cpp file:

[image: image.png]

But I just double checked the version on Github already has these
Serial.println commented out. Maybe you can double check if thats the case
in your version of the library as well. You problem probably has to to this
use Serial somewhere in you code. I guess you can not use the serial port
in anyway if you are using the DCS library.

Update: I just tested locally, and I got the same error when I checkout the
master branch. Had a look at printDualCounter line 58-59 and yes the
Serial.println() statement were there uncommented. Checkout the latest
version on master and the problem disappeared.

So long story short: Download the latest version of the library:
https://github.com/bremme/arduino-tm1637/archive/master.zip and your
problem should be solved.

regards,
Bram

PetitMalin, 2020-06-29:
> No, I was referring to the printDualCounter methods implementation in the SevenSegmentExtended.cpp file: [image: image.png] But I just double checked the version on Github already has these [...]

Wonderfull !!!! 
I tested it works. No more compilation errors. Thanks in any case for your work

bremme, 2020-06-30:
good to hear that the issues is solved. Good luck with your project!

PetitMalin, 2020-07-01:
Thank's again, 

I will write a article (with your Github off course)
Just waiting my Pcbs from Jlcpcb  to take pictures 

Best regards
Gilles