bremme/arduino-tm1637 #42: Upload error

Park-Jinseo, 2020-07-07:
Hello!

Common errors occur in all examples.
The following error occurs when uploading.

What is the problem?

C:\Program Files (x86)\Arduino\libraries\arduino-tm1637-master\src\SevenSegmentFun.cpp:187:80: error: no 'void SevenSegmentFun::bouncingBall(uint16_t, uint16_t, bool)' member function declared in class 'SevenSegmentFun'
 void  SevenSegmentFun::bouncingBall(uint16_t moves, uint16_t d, bool runForever) {

Park-Jinseo, 2020-07-08:
Good! End!

closed 2020-07-08 by Park-Jinseo

felangga, 2020-08-20:
How did you resolve this error ?

Park-Jinseo, 2020-08-20:
> 이 오류를 어떻게 해결 했습니까?

Hello.
You sent me a mail via github.
I don't speak English very well, so I use Google Translate.
So I apologize in advance.

I don't remember the problem right now.
However, looking at other issues, it seems that there was a similar problem.
(https://github.com/bremme/arduino-tm1637/issues/44)

I deleted the library before, so I downloaded it again now.
But the example is fine.

When did you ever download the library?
Did you download it before?
It would be a good idea to download it again now.

If I know exactly how to fix it, I will send you an email again.
Thanks for reading!

Park-Jinseo, 2020-08-20:
> > 이 오류를 어떻게 해결 했습니까?
> 
> 안녕하세요.
> github를 통해 나에게 메일을 보냈습니다.
> 저는 영어를 잘 못해서 Google 번역을 사용합니다.
> 그래서 미리 사과드립니다.
> 
> 지금은 문제가 기억 나지 않습니다.
> 하지만 다른 문제를 살펴보면 비슷한 문제가 있었던 것 같습니다.
> ( [# 44](https://github.com/bremme/arduino-tm1637/issues/44) )
> 
> 이전에 라이브러리를 삭제 했으므로 지금 다시 다운로드했습니다.
> 그러나 예는 괜찮습니다.
> 
> 언제 라이브러리를 다운로드 했습니까?
> 전에 다운로드 했습니까?
> 지금 다시 다운로드하는 것이 좋습니다.
> 
> 해결 방법을 정확히 알고 있으면 다시 이메일을 보내 드리겠습니다.
> 읽어 주셔서 감사합니다!

I am now trying to edit the'SevenSegmentFun.h' file.
I will deliberately write a typo on line 58.

Park-Jinseo, 2020-08-20:
> > > 이 오류를 어떻게 해결 했습니까?
> > 
> > 
> > 안녕하세요. 
> > github를 통해 나에게 메일을 보도록. 
> > 저는 영어를 잘 못해서 Google 번역을 사용합니다. 
> > 그래서 미리 사과드립니다.
> > 지금은 문제가 기억 나지 언어. 
> > 하지만 다른 문제를 살펴보면 더 문제가 있습니다. 
> > ( [# 44](https://github.com/bremme/arduino-tm1637/issues/44) )
> > 이전에 라이브러리를 삭제 했어 지금 다시 다운로드했습니다. 
> > 그러나 예는 괜찮습니다.
> > 언제 라이브러리를 다운로드 했습니까? 
> > 전에 설치 했습니까? 
> > 지금 다시 다운로드하는 것이 좋습니다.
> > 해결 방법을 알고 있으면 다시 이메일을 보내 드리겠습니다. 
> > 읽어 주셔서 감사합니다!
> 
> 이제 'SevenSegmentFun.h'파일을 편집하려고합니다.
> 58 행에 일부러 오타를 작성하겠습니다.

I wrote a typo on purpose and saved it, and the result is the same.
If you have downloaded the library before, there are two ways.

1. Check line 58 of'SevenSegmentFun.h' and correct the typo.
2. Download the library again.

Let me know if it's resolved!

Park-Jinseo, 2020-08-20:
> I am now trying to edit the'SevenSegmentFun.h' file.
> I will deliberately write a typo on line 58.

I made a wrong copy and rewritten.

-------------------------------------------------- -------

Hello.
You sent me a mail via github.
I don't speak English very well, so I use Google Translator.
So I apologize in advance.

I don't remember the problem now.
But if you look at other issues, you can see it.

Now I have downloaded the library again.
I'm trying to edit the'SevenSegmentFun.h' file.
I will deliberately write a typo on line 58.

-------------------------------------------------- -------

I wrote a typo on purpose and saved it, and the result is the same.
If you have downloaded the library before, there are two ways.

1.Check line 58 of'SevenSegmentFun.h' and correct the typo.
2. Download the library again.

felangga, 2020-08-24:
Yes, I solved the problem. 
The library from this git is correct, but the library from the arduino repository still has a typo