bremme/arduino-tm1637 #43: Rotate single digit?

drewsed, 2020-07-08:
Hi,
i use a TM1637to interface a existing display inside a alarm clock, where the first two digits are rotated and the second tow digits are not.

I can drive the display wirth my own code to display the time properly. Now i found your great library and i wonder if it is possible to rotate (180 degree) single digits, so i can use your cool fun section ;) and print functions.

Cheers,
drew

drewsed, 2020-07-08:
Oh, i just realized, that quite along time passed, since here was some activity. I will look into this myself if i find some spare time. 
It is not highest priority in my todo list, though...

bremme, 2020-07-27:
Sorry for my late reply @drewsed. Sounds like a cool addition, although not so many people might actually need it perhaps. But it should not be that hard to implement I think. You are more then welcome to open up a PR. Once I find some time to work on the library again I might implement it myself perhaps.