bremme/arduino-tm1637 #50: Error compiling for Arduino Uno

VaderConsulting, 2021-01-28:
// ==============================INCLUDES=========================================
#include "SevenSegmentTM1637.h"
#include "SevenSegmentExtended.h"

// Date and time functions using a DS1307 RTC connected via I2C and Wire lib
#include <Wire.h>
#include "RTClib.h"
#include <time.h>

#include <MsTimer2.h>
#include <NSEncoder.h>

// ======================================================================
// Display
// Instantiation and pins configurations
// Pin 12 - > DIO
// Pin 13 - > CLK
SevenSegmentExtended display(13, 12);

C:\Users\windo\OneDrive\Documents\Arduino\libraries\SevenSegmentTM1637\src\SevenSegmentFun.cpp:187:80: error: no 'void SevenSegmentFun::bouncingBall(uint16_t, uint16_t, bool)' member function declared in class 'SevenSegmentFun'
 void  SevenSegmentFun::bouncingBall(uint16_t moves, uint16_t d, bool runForever) {
                                                                                ^
C:\Users\windo\OneDrive\Documents\Arduino\libraries\SevenSegmentTM1637\src\SevenSegmentFun.cpp: In member function 'void SevenSegmentFun::print4Bit(uint8_t, uint8_t, uint8_t)':
C:\Users\windo\OneDrive\Documents\Arduino\libraries\SevenSegmentTM1637\src\SevenSegmentFun.cpp:250:65: warning: unused parameter 'y' [-Wunused-parameter]
 void SevenSegmentFun::print4Bit( const uint8_t x, const uint8_t y, uint8_t symbol) {
                                                                 ^
exit status 1
Error compiling for board Arduino Uno.

danielbernalb, 2021-02-03:
I have the same problem with Arduino UNO:
...........src\SevenSegmentFun.cpp:187:80: error: no 'void SevenSegmentFun::bouncingBall(uint16_t, uint16_t, bool)' member function declared in class 'SevenSegmentFun'
 void  SevenSegmentFun::bouncingBall(uint16_t moves, uint16_t d, bool runForever) {
                                                                                ^

danielbernalb, 2021-02-04:
My temporal solution:
Erase SevenSegmentFun.cpp and SevenSegmentFun.h

oum100, 2021-04-26:
I got the same error. This error was wrong typing of "bouncingBall" in file SevenSegmentFun.cpp in line 187.  It not the same as declare in SevenSegmentFun.h.   To fixed it just correct it to "bouchingBall".

schumar, 2021-06-08:
This has already been fixed (in 6f4f084), but you have to manually download the current release ([v1.1.1](https://github.com/bremme/arduino-tm1637/releases/tag/v1.1.1))

B0rax, 2024-06-16:
@bremme can we get a release here? currently the library from the Arduino library manager is broken as it can only install V1.1.0.

bremme, 2024-06-17:
I haven't looked at this for a long time. I just updated the version number in the `library.properties` file to 1.1.1 as well. I guess that should fix the issue right @B0rax ?

B0rax, 2024-06-17:
@bremme i think so, yes. There is also a PR with this fix (#57 )

bremme, 2024-06-17:
Yeah I say that PR as well, but that also included some functional changes. So just made a new commit with the version bump. Gonna close this for now. Let me know if the issue still persists. And thx by the way!

B0rax, 2024-06-17:
@bremme It doesn't look like it was fixed. The Arduino IDE still offer 1.1.0 as the latest release.
I sadly can not add any more expertise here. Maybe it needs to have a release and the update to the `library.properties` at the same time? I don't know.

There is a checklist from Arduino but I guess you know it better than me:
https://github.com/arduino/library-registry/blob/main/FAQ.md#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

bremme, 2024-06-17:
Shit, that's unfortunate. I haven't worked with Arduino for years, so need to check the documentation as well. Let see if I can find some time to fix these issues soon. For the time being you can always just download the library from Github and use it that way ofc.