bremme/arduino-tm1637 #61: SevensegmentFun.cpp

Ardjoen575, 2023-02-26:
#include "SevenSegmentFun.h"

SevenSegmentFun::SevenSegmentFun(SevenSegmentTM1637* display) {
    _display = display;
}

void SevenSegmentFun::counting(uint16_t start, uint16_t end, uint16_t step, uint16_t delayTime) {
    for (uint16_t i = start; i <= end; i += step) {
        _display->print(i);
        delay(delayTime);
    }
}

void SevenSegmentFun::runningLight(uint16_t moves, uint16_t d, bool runForever) {
    uint8_t segs[] = {0x01, 0x02, 0x04, 0x08, 0x10, 0x20, 0x40, 0x80};
    int segLen = sizeof(segs) / sizeof(segs[0]);
    for (uint16_t j = 0; j < moves; j++) {
        for (uint8_t i = 0; i < segLen; i++) {
            _display->setSegments(&segs[i], 1, i);
            delay(d);
        }
        for (int8_t i = segLen - 2; i > 0; i--) {
            _display->setSegments(&segs[i], 1, i);
            delay(d);
        }
    }
    if (runForever) {
        runningLight(moves, d, runForever);
    }
}

void bouncingBall(uint16_t moves, uint16_t d, bool runForever); // function declaration

void SevenSegmentFun::bouncingBall(uint16_t moves, uint16_t d, bool runForever) {
    // function definition goes here
}

Ardjoen575, 2023-02-26:
SevensegmentFun_h

#ifndef SevenSegmentFun_h
#define SevenSegmentFun_h

#include <Arduino.h>
#include <SevenSegmentTM1637.h>

class SevenSegmentFun {
public:
    SevenSegmentFun(SevenSegmentTM1637* display);
    void counting(uint16_t start, uint16_t end, uint16_t step, uint16_t delayTime);
    void runningLight(uint16_t moves, uint16_t d, bool runForever);
    void bouncingBall(uint16_t moves, uint16_t d, bool runForever);

private:
    SevenSegmentTM1637* _display;
};

#endif

Ardjoen575, 2023-02-26:
Compilation error

/sdk/hardware/tools/avr/bin/avr-g++ -no-canonical-prefixes -std=gnu++11 -c -g -Os -w -mmcu=atmega328p -DF_CPU=16000000L -DUSB_VID=null -DUSB_PID=null -DARDUINO=10812 -DIDE="ArduinoDroid" -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO /SevenSegmentTM1637/src/SevenSegmentFun.cpp -o /librariesBuild/SevenSegmentTM1637/SevenSegmentFun.cpp.o /SevenSegmentTM1637/src/SevenSegmentFun.cpp:187:80: error: no 'void SevenSegmentFun::bouncingBall(uint16_t, uint16_t, bool)' member function declared in class 'SevenSegmentFun'
 void  SevenSegmentFun::bouncingBall(uint16_t moves, uint16_t d, bool runForever) {
                                                                                ^
Return code is not 0

B0rax, 2024-06-16:
See #50